sergeykhbr/riscv_vhdl #4: The software models with GUI closes automatically

satya123bora, 2017-02-07:
If I run ./_run_functional_sim.sh or ./_run_systemc_sim.sh, it works fine for 2-3 mins. But, after 2-3 mins, the window closes automatically. The terminal messages are:

[example0]: Plugin post-init example: attr1_='This is test attr value'
[CoreService]:                                                        
**********************************************************
  RISC-V debugger
  Author: Sergey Khabarov - sergeykhbr@gmail.com
  Copyright 2016 GNSS Sensor Ltd. All right reserved.
**********************************************************
[example0]: This is exampleAction(): val=0000cafe                     
<led[7:0]> 01h                                                        
<led[7:0]> 02h                                                        
<led[7:0]> 03h                                                        
<led[7:0]> 04h                                                        
<serialconsole> Boot . . .OK                                          
<serialconsole> ***** BOOTING ZEPHYR OS *****                         
riscv# *** Error in `./appdbg64g.exe': double free or corruption (out): 0x00007f4d8805f1d0 ***
./_run_functional_sim.sh: line 3:  7880 Aborted                 (core dumped) ./appdbg64g.exe -c ../../targets/functional_sim_gui.json "$@"

when the window closes, the terminal also stops working. If I try to run it from a different terminal, it appears for 3-4 secs and then closes. But, If I wait for 15-20 mins and then run it, it again works for 2-3 mins. Why is this happening? I am working on ubuntu 14.04.

sergeykhbr, 2017-02-07:
Thank you for this issue report. I can reproduce it and will try to fix as fast as possible.

sergeykhbr, 2017-02-07:
Bug was fixed. see attribute.cpp, line 216. Allocated memory wasn't initialized (Linux problem only).

closed 2017-02-08 by sergeykhbr