sergeykhbr/riscv_vhdl #5: Rtos port on RISC-V

ninode, 2017-06-22:
Hello,

I'm doing an rtos port on RISC-V and currently working on the fpga-zynq/rocket-chip directory of Berkeley's. 

Could you please provide instruction to how you used Vivado tools to generate another boot.bin which contains another os than linux with the rocket chip ? Did you use [this directory ](https://github.com/ucb-bar/fpga-zynq) at all ?
What kind of files are necessary to have both the rocket cores and another os into the same boot.bin ?

Thank you very much

Regards

sergeykhbr, 2017-06-23:
Hi,

I think you've a bit misunderstood the difference of projects for ZYNQ (ARM+FPGA) and pure FPGA projects. I do not use ZYNQ in my project and I do not use directory you've pointed to. So, I cannot help you to generate boot.bin file because I don't need to generate image for ARM at all.

Regards,
Sergey

closed 2017-06-23 by sergeykhbr

ninode, 2017-07-03:
Hello, 

Actually, is it possible to use your SoC FPGA Rocket sources and push them to a Zedboard to have Rocket cores on the FPGA fabric and not touch the ARM ?

Also, instead of Zephyr, is it possible to use my own sources for a ported RTOS in order to have: RTOS on ARM cores and Rocket cores on FPGA, without the boot.bin, but following your steps ?

Thank you

sergeykhbr, 2017-07-03:
I am not experienced with ZYNQ but I see one common difference in FPGA startup process:

- In a case of pure FPGA project (not ZYNQ), FPGA has access to all HW interfaces and IOs including flash ROM. Software (ISE or Vivado) can directly load bit-image into fpga via JTAG interface either.

- In a case of ZYNQ almost all FPGA's IOs connected to MIO interface between ARM and FPGA. And accordingly with this thread
https://forums.xilinx.com/t5/Zynq-All-Programmable-SoC/Zynq-newbee-questions-Configure-FPGA-fabric-without-hard-ARM/td-p/568025
"**_The Zynq PL cannot be configured independent of the PS; the PL can't power on and read a configuration bitstream from a memory chip. The PS must run and program the fabric through a configuration access port...._**"
In my opinion, it means that you has to build and run at least BootLoader on ARM to properly configure MIO interface and load FPGA image into FPGA.

So, I suppose the answers on questions will be (my opinion only):
1. It is possible to use Rocket sources on FPGA but ARM is always touched.
2. You cannot load FPGA without boot.bin