sergeykhbr/riscv_vhdl #6: compiling errors

Pengnian, 2017-08-17:
Hi ,all

I want to compiling boot (riscv_vhdl\rocket_soc\fw\boot),but find some errors maybe causing  gun-toolchain version , please experts  explanation ! Thanks~

[image]

sergeykhbr, 2017-08-17:
Hi,
You're building _bootimg_ with the flag **-march=rv64ima**, but gcc toolchain was built with the hardware FPU (default configuration).

I suggest to rebuild gcc with _**--disable-float**_ key or build _bootimg_ with the _-march=rv64imafd_.

Pengnian, 2017-08-17:
Okay. Thanks very much，I will  try again！☺️ You explanation  very useful for  me！

On 08/17/2017 17:57, Sergey wrote:

Hi,
You're building bootimg with the flag -march=rv64ima, but gcc toolchain was built with the hardware FPU (default configuration).

I suggest to rebuild gcc with --disable-float key or build bootimg with the -march=rv64imafd.

closed 2017-08-17 by sergeykhbr