sergeykhbr/riscv_vhdl #8: Generate failed of helloworld  hex file

Pengnian, 2017-09-21:
Hi all,

I have used "make"  command in the directory of  "riscv_vhdl\rocket_soc\fw\helloworld\makefiles" want to generate helloworld hex file but the output is all zero.  Besides，the warning cannot solve it !  please tell me the reason. 

[image]
[image]

sergeykhbr, 2017-09-21:
Hi,
you've specified start address (key -b) as 0, so it is possible that generated image at the beginning of file was nullified. Check the linker script *.ld or generated dump/list file to check where the 'code' section is started.

Default reset vector address is equal to 0x1000 and can be changed through the generic vhd-parameter.

Pengnian, 2017-09-25:
@sergeykhbr

 Thank you very much~ 
 It's work~ Thanks your advice!

closed 2017-10-06 by sergeykhbr