sergeykhbr/riscv_vhdl #9: how  to generate bitstream file  by myself ?

Pengnian, 2017-10-11:
Hi Sergey,

At first，thanks for your source code about risc-v cores.

I know I can directly use your  project at ISE or Vivado(riscv_vhdl/rocket_soc/prj/) but I want to synthesis in vivado by myself at xilinx  VCU118 Board. Can  you  tell me which vhdl files  need to add my project and other attention items? 

Thanks you very much !

sergeykhbr, 2017-10-11:
Hi,
I'm doing the following steps to add new fpga/asic target:

1. Create new target ID in the file _techmap/gencomp/gecomp.vhd_ (or reuse '_virtex7_' for your purpose).
2. Generate new PLL module for your target and add it to the file _techmap/pll/SysPLL_tech.vhd_ under your generic parameter. Don't forget to include your pll module into '_techmap_' library (default is '_work_').
3. Do the same step for others '_*_tech.vhd_' files (if needed), like _techmap/mem/bootrom_tech.vhd_ or check that function __is_fpga__('virtex7') returns 1. 
4. Create the constrains file for your fpga
5. Create the target config file (analog is _prj/ml605/config_v6.vhd_) and include it into the project instead of other config target. It should define parameters:
           CFG_FABTECH = virtex7
           CFG_MEMTECH = virtex7
           ..
6. Build project and try to find the words '_blackbox_' or '_black_' in your synthesis output. If not found then OK and everything should be built successfully, otherwise check that you properly included target dependable modules.

Pengnian, 2017-10-12:
@sergeykhbr Hi sergey，
Thanks  for your reply ,I want to ask one question that adding helloworld  hex file to vhdl file synthesis together which attention items we need focus on and why print "hello world !" message  by uart port is messy code. 

please give me  some hint,thanks again!

sergeykhbr, 2017-10-12:
There're used two images:

1. bootimage.hex. This file is generated by fw/boot code. It is valid for ASIC, FPGA and RTL simulation. But there's implemented a special trick (reading 'tech' register from pnp module) for the RTL simulation that allows to skip stage of copying fwimage data from ROM to RAM before executing it.

2. fwimage.hex. It's the image with the firmware (HelloWorld or whatever). In normal mode it should be stored only as a ROM module but to speed simulation SRAM module also is using it for the initialization. This approach normally works on simulation and even for FPGA but invalid for ASIC because SRAM cannot be initialized.

If you have some messy UART output most probably you wrongly setup UART Baud rate. Baud Rate directly depends of Bus clock frequency and UART scaler register must be computed as:

        uart->scaler = Fbus / 115200 / 2
        where Fbus = 40 000 000 Hz
Probably you are using another Bus Frequency.

Pengnian, 2017-10-13:
@sergeykhbr Thanks very much,
It is very important for us !

Pengnian, 2017-10-13:
@sergeykhbr Hi sergey,
Thanks for you patient, I'm very appreciate it !

Pengnian, 2017-10-21:
@sergeykhbr Hi sergey，
 I'm sorry , I can't generate bitstream file for my target board (Virtex UltraScala+ VCU118) following your hints. Can you tell me every steps transplant RISC-V core to my board or create new project in vivado(or ISE)  for my board. My board XDC file   following:

`
##############################################################################
 Generated by PlanAhead 14.7 built on 'Fri Sep 27 19:29:51 MDT 2013' by 'xbuild'
##############################################################################

create_clock -period 5.000 -name i_sclk_p [get_ports i_sclk_p]
set_property IOSTANDARD LVDS [get_ports i_sclk_p]
set_property PACKAGE_PIN G31 [get_ports i_sclk_p]
set_property PACKAGE_PIN F31 [get_ports i_sclk_n]
set_property IOSTANDARD LVDS [get_ports i_sclk_n]

# button "Center"
set_property PACKAGE_PIN L19 [get_ports i_rst]
set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets i_rst]
set_property IOSTANDARD LVCMOS12 [get_ports i_rst]

# DIP switch: SW1.1
set_property PACKAGE_PIN B17 [get_ports {i_dip[0]}]
set_property IOSTANDARD LVCMOS12 [get_ports {i_dip[0]}]
set_property PACKAGE_PIN G16 [get_ports {i_dip[1]}]
set_property IOSTANDARD LVCMOS12 [get_ports {i_dip[1]}]
set_property PACKAGE_PIN J16 [get_ports {i_dip[2]}]
set_property IOSTANDARD LVCMOS12 [get_ports {i_dip[2]}]
set_property PACKAGE_PIN D21 [get_ports {i_dip[3]}]
set_property IOSTANDARD LVCMOS12 [get_ports {i_dip[3]}]

#UART interface
set_property PACKAGE_PIN AY25 [get_ports i_uart1_ctsn]
set_property IOSTANDARD LVCMOS18 [get_ports i_uart1_ctsn]
set_property PACKAGE_PIN AW25 [get_ports i_uart1_rd]
set_property IOSTANDARD LVCMOS18 [get_ports i_uart1_rd]
set_property PACKAGE_PIN BB22 [get_ports o_uart1_rtsn]
set_property IOSTANDARD LVCMOS18 [get_ports o_uart1_rtsn]
set_property PACKAGE_PIN BB21 [get_ports o_uart1_td]
set_property IOSTANDARD LVCMOS18 [get_ports o_uart1_td]

#/ User's LEDs:
set_property PACKAGE_PIN AT32 [get_ports {o_led[0]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[0]}]
set_property PACKAGE_PIN AV34 [get_ports {o_led[1]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[1]}]
set_property PACKAGE_PIN AY30 [get_ports {o_led[2]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[2]}]
set_property PACKAGE_PIN BB32 [get_ports {o_led[3]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[3]}]
set_property PACKAGE_PIN BF32 [get_ports {o_led[4]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[4]}]
set_property PACKAGE_PIN AU37 [get_ports {o_led[5]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[5]}]
set_property PACKAGE_PIN AV36 [get_ports {o_led[6]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[6]}]
set_property PACKAGE_PIN BA37 [get_ports {o_led[7]}]
set_property IOSTANDARD LVCMOS12 [get_ports {o_led[7]}]

# Ethernet signals
set_property PACKAGE_PIN AT22 [get_ports i_gmiiclk_p]
set_property IOSTANDARD LVCMOS18 [get_ports i_gmiiclk_p]
set_property PACKAGE_PIN AU22 [get_ports i_gmiiclk_n]
set_property IOSTANDARD LVCMOS18 [get_ports i_gmiiclk_n]

set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets -of [get_pins igbebuf0/xk7.x1/x1/O]]

set_property PACKAGE_PIN AH30 [get_ports io_emdio]
set_property IOSTANDARD LVCMOS18 [get_ports io_emdio]
set_property PACKAGE_PIN AM31 [get_ports o_emdc]
set_property IOSTANDARD LVCMOS18 [get_ports o_emdc]
set_property PACKAGE_PIN K36 [get_ports i_emdint]
[... 79 lines cut ...]
set_property ALL_PROBE_SAME_MU_CNT 1 [get_debug_cores u_ila_0]
set_property C_ADV_TRIGGER false [get_debug_cores u_ila_0]
set_property C_DATA_DEPTH 65536 [get_debug_cores u_ila_0]
set_property C_EN_STRG_QUAL false [get_debug_cores u_ila_0]
set_property C_INPUT_PIPE_STAGES 0 [get_debug_cores u_ila_0]
set_property C_TRIGIN_EN false [get_debug_cores u_ila_0]
set_property C_TRIGOUT_EN false [get_debug_cores u_ila_0]
set_property port_width 1 [get_debug_ports u_ila_0/clk]
connect_debug_port u_ila_0/clk [get_nets [list pll0/xv7.pll0/w_clk_bus]]
set_property PROBE_TYPE DATA_AND_TRIGGER [get_debug_ports u_ila_0/probe0]
set_property port_width 1 [get_debug_ports u_ila_0/probe0]
connect_debug_port u_ila_0/probe0 [get_nets [list i_uart1_ctsn_IBUF]]
create_debug_port u_ila_0 probe
set_property PROBE_TYPE DATA_AND_TRIGGER [get_debug_ports u_ila_0/probe1]
set_property port_width 1 [get_debug_ports u_ila_0/probe1]
connect_debug_port u_ila_0/probe1 [get_nets [list i_uart1_rd_IBUF]]
create_debug_port u_ila_0 probe
set_property PROBE_TYPE DATA_AND_TRIGGER [get_debug_ports u_ila_0/probe2]
set_property port_width 2 [get_debug_ports u_ila_0/probe2]
connect_debug_port u_ila_0/probe2 [get_nets [list o_uart1_td_OBUF {rst0/AR[0]}]]
set_property C_CLK_INPUT_FREQ_HZ 300000000 [get_debug_cores dbg_hub]
set_property C_ENABLE_CLK_DIVIDER false [get_debug_cores dbg_hub]
set_property C_USER_SCAN_CHAIN 1 [get_debug_cores dbg_hub]
connect_debug_port dbg_hub/clk [get_nets u_ila_1_clkfbout_buf]
`

Pengnian, 2017-10-21:
@sergeykhbr Hi sergey，
I'm sorry , I can't generate bitstream file for my target board (Virtex UltraScala+ VCU118) following your hints. Can you tell me every steps transplant RISC-V core to my board or create new project in vivado(or ISE) for my board. My board XDC file following:
[riscv_soc_vcu118.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1404056/riscv_soc_vcu118.txt)

sergeykhbr, 2017-11-03:
Sorry, I was on vacation.
Where did you stuck? Could you please send synthesis console logs. Because I don't have such board and can't check your constraints file.

Pengnian, 2017-11-06:
@sergeykhbr  Thank you very much,
 "ERROR: [Chipscope 16-213] The debug port 'u_ila_0/clk' has 1 unconnected channels (bits). This will cause errors during implementation."

my synthesis log as follow:

[synthesis_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1446705/synthesis_log.txt)

my implementation log  as follow:
[implementation_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1446708/implementation_log.txt)

Pengnian, 2017-11-07:
@sergeykhbr  Hi,sergeykhbr

Thank you for your patient! 

Vivado synthesis has passed, but implementation report errors , the error is follow:

`[DRC 23-20] Rule violation (INBB-3) Black Box Instances - Cell 'boot0/tech0/genrom1.x0' of type 'boot0/tech0/genrom1.x0/BootRom_vcu118' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
` 
my vivado synthesis log:
[vivado_synthesis_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1448573/vivado_synthesis_log.txt)

my vivado implementation log:
[vivado_implementation_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1448574/vivado_implementation_log.txt)

Pengnian, 2017-11-07:
@sergeykhbr Thank you for your patient!

Vivado synthesis has passed, but implementation report errors , the error is follow:

`[DRC 23-20] Rule violation (INBB-3) Black Box Instances - Cell 'boot0/tech0/genrom1.x0' of type 'boot0/tech0/genrom1.x0/BootRom_vcu118' has undefined contents and is considered a black box. The contents of this cell must be defined for opt_design to complete successfully.`

my vivado synthesis log:
[vivado_synthesis_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1448577/vivado_synthesis_log.txt)

my vivado implementation log:
[vivado_implementation_log.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1448578/vivado_implementation_log.txt)

sergeykhbr, 2017-11-07:
I see at least 2 wrongly synthesized component (see synt. log-file):
```
Report BlackBoxes: 
     +------+----------------+----------+
     |      |BlackBox name   |Instances |
     +------+----------------+----------+
     |1     |BootRom_vcu118  |         1|
     |2     |RomImage_vcu118 |         1|
     +------+----------------+----------+
```
So you need to properly instantiate these modules in a similar way as in *techmap\mem\bootrom_tech.vhd*
1. Check *is_fpga(memtech)* returns true
2. Check that your files included into *techmap* library
3. There's a bit another names than in my project so I can't provide more details.

Pengnian, 2017-11-08:
@sergeykhbr 
Thank for your patient,

My modification steps as follow:

 1. create BootRom_vcu118.vhd for  my target. 
 2.instantiation the modules in `techmap\mem\bootrom.vhd` 
  ( `RomImage_vcu118.vhd` also like above )

So,besides I don't know which file  need to modification .Can you tell  all of `*.vhd` file need to modification  about `BootRom` and `RomImage`.

Pengnian, 2017-11-08:
Thanks for your patient, My modification steps as follow:

   1.create BootRom_vcu118.vhd for my target.
   2.instantiation the modules in techmap\mem\bootrom.vhd
      ( RomImage_vcu118.vhd also like above )
So,besides I don't know which file need to modification .Can you tell all of *.vhd file need to modification about BootRom and RomImage.

sergeykhbr, 2017-11-08:
If you create new ROM-module file then you need to add it into Vivado project and afterward you need to include it into 'techmap' library (default is 'work').

Now it looks like you didn't add your files into the project or didn't include them into proper library by this reason Vivado insert black-boxes.

Pengnian, 2017-11-11:
Okay,Thanks  you very much!!!

closed 2017-11-24 by sergeykhbr