sergeykhbr/riscv_vhdl #11: elf2raw64 и .shstrab

kraziant, 2018-03-29:
Сергей, здравствуйте! Собрали elf2raw64 из исходников. При конвертации выдается ошибка:
 err: undifined .shstrab section.

Возможно мы не правильно elf собираем? 
В качестве тулчейна используем SciFive пресобранный со студией. 

Хотим запустить helloworld.

sergeykhbr, 2018-03-29:
Hi. Could you please attach your elf-file?

Notice: there's bug with some compressed instruction in simulator from the main trunk of the repository that I didn't solve yet.

kraziant, 2018-03-31:
Here they are:

[helloworld.hex.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1865273/helloworld.hex.txt)
[_example-comp.err.txt](https://github.com/sergeykhbr/riscv_vhdl/files/1865275/_example-comp.err.txt)
[helloworld.tar.gz](https://github.com/sergeykhbr/riscv_vhdl/files/1865277/helloworld.tar.gz)

kraziant, 2018-04-02:
Today we a trying to use Ubuntu GNU GCC 6.1.0 toolchain RV64IMA (204MB) and it work fine

sergeykhbr, 2018-04-02:
Thank you for the information. 

I've looked into the code and this message "err: undefined .shstrab section" isn't critical. It only means that sections names weren't found. And it doesn't affect to the final ROM-image.

kraziant, 2018-04-02:
It's look like SciFive pre-build toolchain is use compressed instruction set, which does't support by project cores. In this case using `-march=rv64g` during helloworld compilation, will not solve the problem, because stdlib builded with compressed insturctions.

kraziant, 2018-04-12:
Done

closed 2018-04-12 by kraziant