sergeykhbr/riscv_vhdl #14: How to build River CPU?

mateoconlechuga, 2018-07-13:
Hello,

I understand that this is a template for instantiating the CPU on an SoC, however I am wondering if it is possible to just instantiate the River CPU as a standalone entity? 

I see this issue: https://github.com/sergeykhbr/riscv_vhdl/issues/9

However I am wondering if it is possible to remove any extra components? Thanks for any help!

sergeykhbr, 2018-07-13:
Hi,
I'm not sure that I correctly understood your question, but:

- Yes, you can get River CPU module and connect it to another AXI4 inteconnect. You will also need boot memory, SRAM, interrupt controller, probably Debug Support Unit and other peripheries.
- Or you can generate VCD-trace file from SystemC model of River CPU and use it as stimulus for RTL simulation.

sergeykhbr, 2018-07-13:
Thank you for this question. 

No, I do not use special tool to generate SystemC I do the following steps in development process:
1. Functional models of all devices with integration into SoC functional model. Testing.
2. SystemC model of specific device and automatic comparision of keys parameters relative the functional model.
3. RTL creation and testing relative SystemC using VCD-stimulus.

closed 2018-07-16 by mateoconlechuga