sergeykhbr/riscv_vhdl #15: A few questions

mateoconlechuga, 2018-07-17:
Thanks again for all your help. I just have a few questions concerning the River core:

* It seems that there is just 1 or 2 lines available in the caches. Is there actually a configurable cache size, or is it currently fixed?
* Can RISCV_ARCH be changed to 32 to support the 32bit RISC-V ISA? I see some things that appear to suggest it is hardcoded to 64 bits.

sergeykhbr, 2018-07-18:
1. I've tried to save cache upgrade capability with the possibility to make it more configurable but right now I don't need in it because I don't use DDR memory in my projects. Without DDR the caches are just wasting of resources (in my opinion).

2. No. Initially I had such idea to support RV32 only but now I think it is easy to create the separate branch with 32-bits CPU.

3. I have JTAG Test Access Point (TAP) that should work with this system and I can push it into repository. But I don't want to support gdb by myself in the nearest future.

mateoconlechuga, 2018-07-18:
It would be great if you could add that JTAG TAP whenever you have time; it would be greatly appreciated. Is there any things in particular that need to be done in order to support gdb?

sergeykhbr, 2018-07-18:
I will add JTAG TAP module into vhdl and update doxy with the description of scan chain values but I have very limited knowledge of the OpenOCD protocol. I suppose there should be configuration file or front-end module to translate gdb string commands into rd/wr addr/value transaction.

Current doxygen already contains information about DSU registers. Additional information of how-to-use them can be found in debugger's source code.

P.S. See https://www.gaisler.com/products/grlib/grip.pdf, page 66, "JTAG Debug Link" for additional information. The difference only in address width 64 bits instead of 32.

mateoconlechuga, 2018-07-18:
Great, thanks for your help!

sergeykhbr, 2018-07-28:
I've added the JTAG TAP into the system top-level and made simple initial tests in rtl simulator.

closed 2018-07-28 by sergeykhbr