sergeykhbr/riscv_vhdl #16: Boot procedure bypass ROM FW copy to SRAM

daffy1108, 2018-07-18:
Hi Sergey,

Thanks again for making this RISCV design open to design community.

I'm trying to add a new peripheral and add programming.

So I modified the ADDRESS MAP and logic to add this peripheral.

Is there a shortcut way to add my new code right after

"Registers initialization"

and before

"Copying ROM FW Image into internal SRAM" 
"Go to SRAM entry point 0x10000000 in user mode"
"Initialization of the "Interrupt Registers" by proper handler"
"Initialization of the UART"
"Start main task"
?

The sample code also has GPIO writing to the LED.
How to make new code also work before or after GPIO writing to LED.

For simulation, the ROM FW Image copying into internal SRAM is consuming lots of
simulation time and doesn't make for quick and easy debug of my code and logic.

Any other recommendations ?

Thanks,

David

sergeykhbr, 2018-07-18:
- Download the latest GCC from https://github.com/gnu-mcu-eclipse/riscv-none-gcc/releases/ (thanks to Liviu Ionescu <ilg@livius.net>).
- Build elf2raw64 tool and copy it into gcc/bin folder (setup PATH variable).
- Edit examples/boot/src/main.c file by adding your functionality into method *_init()* before *copy_image()* call.
- Modify method led_set(int) accordingly with your requirements.
- Run make (or make.bat) from *examples/boot/makefiles*.
You should get the new HEX-file with your modifications directly used by rtl.

You can do the same things with other examples. 

Image is copying under conditions in bootloader:

        if (tech != TECH_INFERRED && pnp->fwid == 0) {
            memcpy(sram, fwrom, FW_IMAGE_SIZE_BYTES);
        }
It means:  not behaviour simulation AND run the first time after power-on. So I suppose you are running NOT behaviour simulation.

daffy1108, 2018-07-18:
Hi Sergey,

Looks like we can modify
fw/boot/src/main.c
to add the peripheral code before "copy_image()".
If we were to comment out the "copy_image()" would RISCV read and execute 
from ROMIMAGE instead of SRAM ?

void _init() {
    uint32_t tech;
    pnp_map *pnp = (pnp_map *)ADDR_NASTI_SLAVE_PNP;
    uart_map *uart = (uart_map *)ADDR_NASTI_SLAVE_UART1;
    // Half period of the uart = Fbus / 115200 / 2 = 70 MHz / 115200 / 2:
    //uart->scaler = 304;  // 70 MHz
    uart->scaler = 260;  // 60 MHz

    led_set(0x01);
    print_uart("Boot . . .", 10);
    led_set(0x02);
///////////////////////////// PUT NEW PERIPHERAL CODE HERE ////////////////
    copy_image(); /////////// WHAT HAPPENS IF WE COMMENT THIS OUT ???? //////
    led_set(0x03);
    print_uart("OK\r\n", 4);
    led_set(0x04);
}

closed 2018-07-25 by sergeykhbr