sergeykhbr/riscv_vhdl #17: loadelf is not working

southlife, 2018-07-24:
Hi, sergey.

Thank you for making River system.
I am following the tutorial now.
However, there are parts that do not work so I leave a question.

[loadelf helloworld] command is work well in Simulator(_run_functional_sim.sh).
But, in the FPGA(_run_fpga_gui.sh), the loadelf command is not work.

Here is the log when I run the command.

> riscv# loadelf /test/helloworld                           
> riscv# err: undefined .shstrtab section
> [loader0]: Reading '.text' section                                    
> [loader0]: Reading '.rodata' section                                  
> [loader0]: Reading '.rodata.cst16' section                            
> [loader0]: Reading '.rodata.str1.8' section                           
> [loader0]: Reading '.data' section                                    
> [loader0]: Reading '.got' section                                     
> [loader0]: Reading '.got.plt' section                                 
> [loader0]: Reading '.sdata' section                                   
> [loader0]: Reading '.sbss' section                                    
> [loader0]: Reading '.bss' section                                     
> [loader0]: Loaded: 33808 B                                            
> [edcltap]: ../src/libdbg64g/services/debug/edcl.cpp:154 No response. Break write transaction.
> [edcltap]: ../src/libdbg64g/services/debug/edcl.cpp:154 No response. Break write transaction.
> [edcltap]: ../src/libdbg64g/services/debug/edcl.cpp:154 No response. Break write transaction.
> [edcltap]: ../src/libdbg64g/services/debug/edcl.cpp:154 No response. Break write transaction.
> [edcltap]: ../src/libdbg64g/services/debug/edcl.cpp:154 No response. Break write transaction.

As you can see, when I run loadelf, edcl does not work properly.
Does it work right in the FPGA?

sergeykhbr, 2018-07-24:
I will check how the latest modifications affected on FPGA target. Did you use the latest version?

Could you also make the following steps to get additional information:
- Halt FPGA before loading new image (because of when loading is finished *loadelf* command restores CPU previous state).
- Check that *npc* register is equal to RESET_VECTORE (value 0x40)
- Read or dump part of memory starting from 0x10000000 (in hex format) and compare it with the generated *helloworld.hex* file.

Thank you

sergeykhbr, 2018-07-24:
I've just tried to build and debug FPGA (ML605) target using Ethernet/UDP TAP and it works fine. The only changes that I've made were the following things:
- Add recently created  *dcom_uart.vhd* file into *misclib* of the ISE project.
- I also wrote 1 into **pnp->fwid** register by sending command 'write 0xfffff004 4 0x1' before loading elf-file. It allows to avoid default image (zephyr) re-loading after soft-reset (see boot/main.c).

Could you provide more information about your setup?

southlife, 2018-07-25:
Thanks for your fast response.

I had test with latest pre-built image(riscv_vhdl-master\rtl\bit_files\v6.0\virtex6_ml605\riscv_soc_river_40mhz_zephyr16_v6.bit) and ML605 FPGA board.

1. When I halt the target before loading elf file, NPC value is 0x10004258 not 0x40.
2. Write 0x1 to 0xfffff004 for bypassing image_copy function in boot-up time as you mentioned.
3. And run [loadelf helloworld] command
4. Then UART0 display the TRAP code as following.

> shell> Boot . . .OK
> ADC clock not found. Enable DIP int_rf.
> mcause:0000000000000002,mepc:0000000010000000

But when I use the latest rtl code(rebuild fpga image in ISE) that you have updated yesterday(added dcom_uart.vhd, dcom_jtag.vhd, tap_jtag.vhd), 
It works fine as following!

> Boot . . .OK
> ADC clock not found. Enable DIP int_rf.
> Hellow World - 1!!!!
> mcause:0000000000000002,mepc:00000000000000de

I think it would be a good idea to add writing 0x1 at 0xfffff004 to the tutorials

sergeykhbr, 2018-07-25:
Thank you for your feedback.
I'll move initialization of register *pnp->fwid* into the boot code.

closed 2018-07-25 by sergeykhbr