sergeykhbr/riscv_vhdl #18: riscv_soc_tb crashes in Vivado Simulator

mateoconlechuga, 2018-07-25:
When I execute the simulation from Vivado 2017.4, in the KC705 project, it produces the following errors:

    Command failed: Simulator command interrupted.
   
    Simulator engine not responding.
    The simulator has terminated in an unexpected manner. Please review the simulation log (xsim.log) for details.

I'm not sure what may be the cause? Does this occur on your end?

sergeykhbr, 2018-07-25:
I've never run Vivado simulator before. I'm using QuestaSim and the unisim libraries to simulate fpga targets.
Recently I've built and commit new bit-file for KC705 board with all the latest modifications you can try it on your board.

mateoconlechuga, 2018-07-25:
Oh I see thanks!

closed 2018-07-25 by mateoconlechuga