sergeykhbr/riscv_vhdl #24: freertos port for river core

suhasskrishanmurthy, 2019-06-14:
Hi, I want to port freertos on the river core,so is there any links where i can refer or if you have done it would you share the steps? . I have gone through the freertos site

sergeykhbr, 2019-06-14:
Hi,
I didn't port freertos and I don't have necessary steps. 

If you are not going to change ROM bootloader implemented in the SoC you should build freertos as a usual user application with program entry point 0x10000000 (internal SRAM base address) and change parameter CFG_SIM_FWIMAGE_HEX in *config_[target].vhd*.

Bootloader will copy your application image from ROM with this image into internal SRAM and jump to the entry point.

To implement serial output/input (printf/cin) please refer to the examples in folder *./examples*

suhasskrishanmurthy, 2019-06-17:
thank you for replying.
I actually went through this link https://www.freertos.org/Using-FreeRTOS-on-RISC-V.html
is this sufficient or any changes to be made? 
I just have a few questions.
1.is there any changes to be made for memory map,interrupts tables,uart etc..

sergeykhbr, 2019-06-18:
Your question is too common.
Freertos tutorial has a statement that 3 boards are supported: MiFive, VEGABoard ans SiFive. So if you want another one you need to add it manually changing configuration and source files, memory map, interrupts, drivers etc. I'm not going to port freertos myself in the nearest future.

If you have problems with my rtl code or simulator please attach error message and log files.

closed 2019-06-18 by sergeykhbr