sergeykhbr/riscv_vhdl #26: What configuration did you use for Rocket core?

teadotjay, 2019-07-19:
Version 7 and below included a pre-built Rocket core. Can you make available the configuration files for this version? I'm guessing you customized it a bit before generating from the Chisel code.

Thanks,
Teadotjay

sergeykhbr, 2019-07-31:
Yes, I'd been customizing Rocket because of another memory map at least. I used a set of hacks that shouldn't be used during rocket-chip configuration so it is better to ask in a thread or via hw-dev forum with related project.

closed 2019-07-31 by sergeykhbr

teadotjay, 2019-08-06:
Not sure I understand the answer. I'm building a platform to compare different RISC-V implementations, and yours seemed a convenient place to start since it includes both Rocket and River. However, I want to know what changes you made from the original Rocket. That seems too specific to ask on SO, but I can if you think that's appropriate.