sergeykhbr/riscv_vhdl #29: Openning ttyUSB0 at 115200 . . .failed"

suhasskrishanmurthy, 2019-08-29:
I am trying to run application debugger on top of fpga.
so i executed fpga_gui.sh and used loadelf tool to load the zephyr.elf onto fpga but it says 
[0, "port1", "../src/libdbg64g/services/comport/com_linux.cpp:152 fopen() failed"]
[0, "port1", "../src/libdbg64g/services/comport/comport.cpp:128 Openning ttyUSB0 at 115200 . . .failed"]
[0, "edcltap", "../src/libdbg64g/services/debug/edcl.cpp:99 No response. Break read transaction[1] at 80090000"]
[0, "udpedcl", "../src/libdbg64g/services/debug/udp_dbglink.cpp:224 sendto() failed

this never stops
and also LED @ ethernet do not blink at host system side 
i have used sudo  ifconfig eth0 192.168.0.53 netmask 255.255.255.0
to change the ip of my pc 
So please tell what am i doing wrong?

sergeykhbr, 2019-08-30:
I think there're at least two problems:
   - Serial output doesn't work
   - No debug connection via Ethernet

To start solving the 1-st one you should load any fpga image that can generate uart output (river soc or anything other). If you would see correct output in any console client then it should work in simulator's 'serial console' widget when you specify the same port settings in JSON-configuration file (*fpga_gui.json*).

The second problem may be caused by firewall settings or something like that, for me it is harder to help because I don't use linux for the hardware debugging. Maybe you will find something helpful in this document (chapter 5)
[riscv_vhdl_trm.pdf](https://github.com/sergeykhbr/riscv_vhdl/files/3559007/riscv_vhdl_trm.pdf)

FrankAnze, 2019-09-20:
Hi, I had the same problems.

I solved uart's problem by changing 'COM3' to '/dev/ttyUSB0' ('ttyUSB0' doesn't work for me) but I couldn't solve the debug connection via ethernet.

I can see the FPGA from my PC as you can see in the following image:

[image]

I noticed that when running './_run_fpga_gui.sh' an error related to CoreService 'core0' appears at the beginning:

[image]

While running the simulator './_run_functional_sim.sh' the following appears:

[image]

I was also seeing the files 'fpga_gui.json' and 'functional_sim_gui.json' and I noticed that 'fpga_gui.json' has much less services than 'functional_sim_gui.json' such as 'TcpServerClass', 'SerialDbgServiceClass', 'CpuRiver_FunctionalClass'.
I didn't find any service called 'core0' in 'fpga_gui.json' but I have no idea if that could be the problem. Could it be that there is a problem with 'fpga_gui.json'?

Regards,
Frank

sergeykhbr, 2019-09-21:
This message formed by *ConsoleClass* and used to debug zephyr os during development. I removed this attribute and you won't see it anymore.

This modification cannot affect Ethernet debug link. Probably you could try to find information about how to setup EDCL debug link on linux for LEON3 boards because they use absolutely the same ethernet module and settings. 
Take into account that EDCL module of MAC can response only on ARP requests and ignore all others. By this reason firewalls can mark the device as unhealthy or something like that. I disable all services and protocols except IPv4 for the eth. port connected to fpga board (on windows)

closed 2019-11-26 by sergeykhbr