sergeykhbr/riscv_vhdl #30: Generating bit file for boot up using debugger

taniyagarg4, 2019-09-05:
I am using a linux machine for fpga implementation of river core with freertos.
As you suggested, i changed COM3 to ttyUSB0 in debugger/targets/fpga_gui.json
After this, when i tried to generate bit file, i got the error that it can not access bin folder in riscv_vhdl/examples/dhrystone/makefiles/
So, i checked the repository and saw that the folder name was changed from bin to binarm
I renamed the folder to bin
As per your suggestions, I also did the following:
      1. desabled FPU
      2. commented CFLAGS += -march=rv64imac -mabi=lp64 statement in examples/bootrom_tests/makefiles/make_bootrom_tests

After this, when I tried to generate the bit file, it gave the error that I am trying to access the dhrystone21.hex after the limit of the file.
After further debugging, I found that the ROM_LENGTH calculation in rom_inferred.vhd file exceeds the dhrystone.hex file limit.
Can you please check these issues and let me know what am I doing wrong?

sergeykhbr, 2019-09-05:
Just several comments because in your question actually several questions:

- You can freely change json-configuration files it cannot affect the build process.
- *examples/dhrystone/makefiles/binarm* folder contains the **pre-built** hex-image + elf (with debug symbols) for ARM processor (not a RISC-V). These files are used by target *functional_arm_gui.json* or with the custom ARM-R5 system that not in the repository. You shouldn't rename it. 
- FPU is disabled by default for *boot/* example. See the parameter *FPU_ENABLED=0* in the *make_boot*. Total size of the boot image 3 KB it is much less than available 'bootrom' space.
- I am using own utility **'examples/elf2rawx'** to generate hex-file. So you can try to build it and copy the binary file 'elf2rawx' into the folder with your gcc compiler:

         $ cd examples/elf2rawx/makefiles
         $ make
         $ cp elf/elf2rawx $(YOUR_CUSTOM_RISCV_GCC_FOLDER)
         $ cd ../../dhrystone21/makefiles
         $ make

- What version of gcc you are using (or actually where I can get the same to reproduce your result)?
- Are you going to port freertos yourself?

taniyagarg4, 2019-09-05:
i have actually followed your steps and sorry we are booting zephyr the way
you have done.
i was to change the folder name because it is giving an error while
generating bit file. So if i should not rename then what is to be done

sergeykhbr, 2019-09-05:
I think I've understood. There're special *.vhd configuration file for each project:
- *rtl/prj/kc705/config_k7.vhd* for the KC705 board
- *rtl/prj/ml605/config_v6.vhd* for the ML605 board
- *rtl/prj/modelsim/config_msim.vhd* for the simulation in QuestaSim
- etc

All these files have the following constants definition:

    constant CFG_SIM_BOOTROM_HEX : string := 
                CFG_TOPDIR & "examples/boot/linuxbuild/bin/bootimage.hex";

    constant CFG_SIM_FWIMAGE_HEX : string := 
                CFG_TOPDIR & "examples/zephyr/gcc711/zephyr.hex";

Make sure that these constants are the same as shown above to run zephyr. It possible to change them to run bare-metal Dhrystone benchmark or *bootrom_tests* for CPU verification or redefine *CFG_TOPDIR* parameter.

taniyagarg4, 2019-09-06:
Ok sir I will try this but if I want to use the hex image I generated using
your specified method, then what should I do because that is the one that's
giving error?

sergeykhbr, 2019-09-15:
In that case please share information about gcc version and build log.

closed 2019-09-17 by sergeykhbr