sergeykhbr/riscv_vhdl #35: Communicating with the ML605 Board (v8.0)

gutkedu, 2020-03-11:
Hello,

I'm having an issue or misunderstanding for communicating with the processor via UART. I downloaded the bitstream v8.0 into the ML605 board using iSE Impact, opened the UART channel on my computer and when I press RESET, I receive only a byte 00 on UART. 

Also, it is possible to generate the bitstream for the ml605 board in the v11?

Thanks

sergeykhbr, 2020-03-12:
Hello,
- Check that you setup 115200 baud rate.
- ML605 is a quite obsolete board which is not supported in Vivado (I entirely switched into kc705). But you can build bit-file in ISE Studio from the latest 'master' (I've just checked it) and try to run it on your board.

gutkedu, 2020-03-13:
I checked and it is on 115200 baud rate, I tried multiple versions of the bitstream already, version 7.0, 8.0 and 11.0  (generated the bitstream with Xilinx ISE myself) but no success, I`m still receiving  0x00 on uart input. Maybe my board has a problem...

sergeykhbr, 2020-03-13:
Did you try to connect to board via Ethernet?
Could you try the following file?
[r5_soc.zip](https://github.com/sergeykhbr/riscv_vhdl/files/4331381/r5_soc.zip)

gutkedu, 2020-03-13:
I didn't try connecting with the ethernet and downloading the debugger. I tried this r5_soc bit file and still the same issue, next week I'm going to try to find the problem again, could be a problem with the board or with the software, I'm not sure

closed 2022-09-15 by sergeykhbr