sergeykhbr/riscv_vhdl #41: How to generate VCD files of bare-metal dhrystone with _run_systemc_sim.sh?

cnjsdfcy, 2021-09-27:
Hi Sergey,

Thanks for sharing this nice project.

I assume RISC-V SystemC model is CABA(Cycle-Accurate Bit-Accurate), right?

And I want to view it's waveform of some workloads, e.g. the pre-built dhrystone21.elf.
So, I managed to build the master branch code successfully (some modifications to satisfy compilation and added some missing files in the makefile), then I tried to add loading dhrystone21.elf with below modification:
[image]
But, after waiting some seconds, seems nothing have been run, please check the **attachment for log**.
[sc.log](https://github.com/sergeykhbr/riscv_vhdl/files/7236011/sc.log)

And below is screenshot of the pop-up Universal platform simualtor:
[image]

Any suggestions to correctly dump VCD waveform?

Thanks,
Cai Yu

sergeykhbr, 2021-09-27:
Hello, your assumption about the cycle accurate model is correct.

Here's the short description of the JSON-configuration file:
1. 'InitCommands' list attribute contains a list of commands that will be send to simulator during startup. Command 'loadelf _filename_ <b>nocode</b>" just loading symbol information (available in the Symbol Browser of simulator) without actual code flashing.
2. File image that actually used during CPU boot is defined here:
[image]
3. To generate VCD file just define output file name here:
[image]
4. Then you will be able to use any viewer (gtkvave as an example):
[image]

cnjsdfcy, 2021-09-27:
Thanks for your quick reply.

I tried with the bootrom_tests.hex, but it still not running correctly:
[image]

sergeykhbr, 2021-09-27:
Yes, I'm developing new updates for the CPU and disconnect temporary interrupts request pins from pipeline. This problem is a sequence of current development because bootrom_tests intentionally reads/writes into unmapped regions to check proper exceptions handling. You could try stable tagged version v11.0 to run this test.

cnjsdfcy, 2021-09-27:
I tried both v11.0 and v10.0, both warning with access of unmapped address, but V10.0 print out _HARTID_ in uart0 quickly however V11.0 doesn't. Anyway, I will try to modify and build the bootrom_test to generate a clean and short test/waveform.

Thanks, I am closing this issue.

closed 2021-09-27 by cnjsdfcy