sergeykhbr/riscv_vhdl #45: Adding a slave peripheral

Mina2411, 2023-05-20:
I want to add a new slave peripheral to Riscv, and I took SRAM as reference to know the modules and packages that I should modify, and I found that I should add:-
1) Index, start and end address in the memory address mapping in types_bus0_pkg.
2) Add a new device id in types_amba_pkg for my new peripheral.
3) Define the size of the peripheral in config_target_pkg.
4) Define the new slave peripheral in riscv_soc_pkg which I can't understand what this number given to the peripherals represent and to which module this number is used "I think this is the missing part that this number should be mapped to another module and get some information from this module, so if you can clarify what is the idea behind this number "SOC_PNP_SRAM"". 
Do I need to adjust any other modules or packages?

sergeykhbr, 2023-05-20:
1. Yes. You need to create new AXI slave slot by increasing CFG_BUS0_XSLV_TOTAL value and defining allocated memory range in the CFG_BUS_MAP.
2. It is optional. Only if you would like to use plug'n'play feature (device descriptor visible in pnp module). You can use any values here including zeros. This IDs do not affect hardware.
3. It is optional. config_target_pkg.sv file should be used to specify parameters that could have different value for different targets. For an example, internal SRAM in ASIC target should have 2 MB and it uses full allocated memory space in CFG_BUS_MAP but it is too big memory for FPGA board kc705. You can hardcoded parameters of your module wherever you want.
4. It is optional the same as p.2. Only if you want that descriptor of your module was available in pnp module. You may ignore this structure and output signals of the module.

The best approach is to make step 1 and use _axi_slv_ sub-module as an AXI interface inside of your device.

Note: Most of the Verilog and SystemC files are generated by __rtlgen__ utility (see https://github.com/sergeykhbr/rtlgen). When it will be fully finished it will implement full system generator that allows to create topology of the SoC and full customization.

Mina2411, 2023-05-20:
Hello,
you mean to create a slave peripheral and fit it in the riscv soc, I need to do all the modifications in types_bus0_pkg only which is the index given to the peripheral to access the memory mapping vector depending on the size of the peripheral of course and increase CFG_BUS0_XSLV_TOTAL value. then I need to take the axi_slv as a sub module which I did already, anything else is optional which I may or may not use right?
Thanks

Mina2411, 2023-05-20:
I see also that there is no axi_master interface or it is implemented in the sv version under a different name, because I wandering how the master peripherals are connected to the AXI without an interface module, so can you clarify where is the AXI/Master interface module?

sergeykhbr, 2023-05-21:
Master interface is in the development to provide coherence interface and re-used in CPUs workgroup

Mina2411, 2023-05-21:
so you mean that there is no module for AXI_master interface explicitly, sorry,I am not able to understand what you mean .

Mina2411, 2023-05-21:
As I want to add a new master peripheral, how can I communicate with the AXI bus? That's why I am asking about an interface module like the axi_slv.

sergeykhbr, 2023-05-21:
I cannot provide simple solution right now. You have several options that should be implemented manually:

1. Connection to ACP port (before L2-cache). If you are developing DMA engine or another CPU and would like to provide coherence of caches of your device and CPUs of the workgroup then you need to reproduce state machine similar to _river_amba.sv_ file.
2. Connection to BUS0 master port (no hardware coherency). Simplified version of the _river_amba.sv_. In this case you either have two options: (a) Properly configure PMP (or MPU - memory protection unit) that allocates uncached memory regions used for inter-processor communication or (b) actively use ifence instruction and loose performance.
3. Implementation of the master interface module at the top of my todo list, so sooner or later I will commit such module.

Mina2411, 2023-05-31:
Hello again,
I mas

> 1. Yes. You need to create new AXI slave slot by increasing CFG_BUS0_XSLV_TOTAL value and defining allocated memory range in the CFG_BUS_MAP.
>
>     2. It is optional. Only if you would like to use [...]

Hello again,
I made all the steps, and I defined the memory map for a slave peripheral as '{64'h0000100000000, 64'h0000100000100}, and The index for the peripheral to access the memory map, but when I test the peripheral by writing a value and trying to read it again to see if the peripheral is fitted in the processor or not, it always fails.I tried the same implementation of the sram to see if the problem is in my implementation or not, and it fails also. so Is there any other software "I mean in C header files or C files" or hardware steps that I should follow or the previous steps should let everything work?

sergeykhbr, 2023-06-01:
- Could you share the code snippet of how you are connecting your module to AXI bus?
- Are you able to run RTL simulation to check the access?

Mina2411, 2023-06-01:
[axi4_new_slave.txt](https://github.com/sergeykhbr/riscv_vhdl/files/11623343/axi4_new_slave.txt)
For the testbench, I am working on it, but as you can see I made the connection the same as the connection of the sram and any other slave module.
Thanks

sergeykhbr, 2023-06-01:
I mean at the SoC level (file riscv_soc.sv) not inside of the slave device.

Mina2411, 2023-06-01:
[riscv_soc.txt](https://github.com/sergeykhbr/riscv_vhdl/files/11623663/riscv_soc.txt)
This is riscv_soc module, you can find the new peripheral under the sram module port map.

sergeykhbr, 2023-06-01:
That looks good for me. But I think you should increase upper address. You cannot allocate less than 4 KB of memory on AXI bus (it is possible on APB - bus1).

Mina2411, 2023-06-02:
Hello,
Yes,that was the problem, it worked now.
also, I need to make my new peripheral non-cachable, so what I did that I tried to configure some CSR registers in the software program, and I did the following:- 

1) configure CSR_MPU_ADDR_Register and give it the value of the base address of my new peripheral. 
 __asm__("csrw 0x352, %0" : :"r"(base_address));

2) configure CSR_MPU_Mask_Register and give it the size of the peripheral.
 __asm__("csrw 0x353, %0" : :"r"(region_size));

3) configure CSR_MPU_CTRL_Register to adjust the attributes needed for the new peripheral 
__asm__("csrw 0x354, %0" : :"r"(0x10011)); //enable region/uncachable region/not an executable region/read region/write region

Actually, I added these lines of code in the main function, and it didn't work, so is there something else to be done to disable cache access when I am trying to read from or write to the peripheral?

Thank you very much for your effort.

sergeykhbr, 2023-06-02:
See Physical Memory Protection (PMP) in risc-v specification. I forget to remove _map_mpu.h_ header file from examples that you are referring to. It was my own version of protection module before the specification updates and it is obsolete.

Mina2411, 2023-06-02:
You mean that I need to configure registers for PMP in the software program ?

sergeykhbr, 2023-06-02:
No, sorry. My bad. PMP setups access rights. Edit _pma.sv_ file to specify new IO (uncached) region.

Mina2411, 2023-06-02:
I saw the pma module, and I didn’t understand what I should do to specify uncached memory regions, or I need to add something to the module itself

sergeykhbr, 2023-06-02:
In pma_pkg.sv specify constants:
```
localparam bit [CFG_CPU_ADDR_BITS-1:0] YOUR_DEVICE_BAR = 48'h000100000000;
localparam bit [CFG_CPU_ADDR_BITS-1:0] YOUR_DEVICE_MASK = 48'h000000000fff;// 4 KB
```
In pma.sv add uncahced data region:
```
    end else if ((i_daddr & (~YOUR_DEVICE_MASK)) == YOU_DEVICE_BAR) begin
        v_dcached = 1'b0;
```

Mina2411, 2023-06-04:
Hello, 
I tried to run the test bench of kc705_top, and I tried to monitor some internal signals that should change according to the software program that initialize the rom, but the signals are not assigned to any value for a long simulation time. does the test bench implemented to test how the program will affect the signals "such as signals on the AXI bus", or to monitor other features?
Thanks.

sergeykhbr, 2023-06-05:
I am not sure that I understood your question but asic_sim, kc705_sim and a real hardware execute absolutely the same code and should show the same output. With the enabled DDR controller kc705_sim simulation works mush slower than 'unisim' asic_sim.

CPU harts after reset wait until I/D caches finish its initialization and become available. Duration of the cache initialization depends of cache sizes.

Mina2411, 2023-06-05:
I mean that I want to test the peripheral in vivado simulator (xsim) using the software program that initialize the ROM, and I am trying to monitor some internal signals that should change depending on the software program, but the problem is that the signals aren't changing, so my asking that should the test bench of the top module simulate all the submodules, and I should see how the AXI signals are changing.

sergeykhbr, 2023-06-05:
Yes, whole system are simulating with the all peripheries.

Mina2411, 2023-07-14:
Hello sergey,
I have a problem that whenever I try to add another peripheral on bus0 in this address range '{64'h0001008000000, 64'h0001008200000}, the simulation crashes, and bitstream also doesn't work, so is there a reason for that or something I did wrong.
Thanks

Mina2411, 2023-07-14:
Hello Sergey,
I want to add another peripheral on bus0 in the memory region '{64'h0001008000000, 64'h0001008200000}, and when I tried to do so, the simulation crashes and bit stream isn't working, I think it something related to axictrl_bus0.sv module, but I don't know what is the exact problem to change it. I am also working on the old version of axictrl_bus0 not the new one that you committed recently, so if you can tell me what to change in the old version because by using the new one this will lead to many changes in the previous steps that I made.
Note:- This peripheral is of index 7, so it is number 8 on the bus.
Thanks

Mina2411, 2023-07-14:
Hello Sergey,
I was trying to add another peripheral on bus0 in the memory region '{64'h0001008000000, 64'h0001008200000}, and after this modification in bus0_pkg, the simulation crashes and bitstream doesn't work as well, so is there a reason for that or I made something wrong knowing that this peripheral of index 7, so I have 8 peripherals on bus0 "localparam int CFG_BUS0_XSLV_TOTAL = 8;"
Thanks

sergeykhbr, 2023-07-23:
You can use fence.i instruction to flush caches