sergeykhbr/riscv_vhdl #46: simulation of kc705_tb

mito200, 2023-05-31:
Hello Serge,
You have done great work in this project. I really appreciate it. However, I am now stuck in the simulation step of kc705 testbench. I followed these steps to simulate it. I have gone to kc705_sim directory and then run make build command and then make gui but it is not responding. It is displaying this message "make: srun: No such file or directory". Do you have any idea how to get it working? 
Thanks in advance.

sergeykhbr, 2023-05-31:
Hi,
- srun is the executable binary of the Slurm utility. It is the resource manager used by a computer cluster. If you intend to run makefile on a single computer you need to remove srun and related arguments.
- To run kc705 simulation "as is" you need to know how to link Secure IP (SIP) library with you simulation utility. Xilinx doesn't provide SIP source code. KC705 DDR controller contains a lot of SIPs. Otherwise, _asic_sim_ target uses only unisim components.
- I think the easiest solution for you will be to add SystemVerilog source and constrain files from folders (with sub-folders) _sv_, _kc705_ and _kc705_sim_ to an empty Vivado project and use ISim simulator for simulation.

mito200, 2023-05-31:
I have tried this solution but the simulation window is not even opening . It crashes from the beginning. Are there any certain configurations for the simulation in the make file? In addition, I have nullified the DRAM as I will not use it. Hence, I removed everything related to it in the testbench.

sergeykhbr, 2023-06-01:
Not enough information. Not enough memory /CPU resources or something wrong with the port.

sergeykhbr, 2023-06-08:
Looks good for me at this point. I've never tried to use ISim, do you see any messages in the vivado console or triggered breakpoint?

mito200, 2023-06-09:
Unfortunately, No

mito200, 2023-06-09:
I think trying the make file is a good solution . However, To remove srun from the make file , Can I just comment all lines containing SRUN command. I am not familiar with srun actually , so it is really appreciated if you can help.

mito200, 2023-06-09:
Moreover, I have nullified DDR component . Therefore , I think I do not need to link SIP with my simulation , right?

sergeykhbr, 2023-06-12:
- Try to check input ports _i_xmsto_ and _i_xslvi_ of the _axictrl_bus0_ module that there're no 'X' values.
- Try to update to the latest version, some bugs were fixed on pnp-bus. They do not stuck the ncsim simulation but ISim could be more sensitive to that types of bugs.

mito200, 2023-06-12:
I have just used it but it is the same output
I think the problem is in cache as in my program v_resp_valid in instruction_cache is not set to 1 at all. I think this will affect the whole simulation . I do not know where the problem is actually but I think it must be set to 1 so that the fetch proceeds.

sergeykhbr, 2023-06-15:
Could you try to update module axi2apb_bus1.sv file and try again? There was bug detected with the array index.

mito200, 2023-06-27:
Actually, I have tried it and it is not working also , Serge

mito200, 2023-06-27:
Can you please try to simulate it on xsim , So that I know that the problem is not in my environmnent?
Thank you for your efforts

mito200, 2023-06-30:
Hello Serge again,
I have found the bug that stopped the simulation. Here ctrl_path_id was displaying x value for me in simualtion and this causes the simulation to stop before filling first cache line . Hence , I think the problem is that vivado is considering ctrl_path_id=0 line to be another driver to ctrl_path_id . So, I have commented this line as shown and it is working right now . I know that it is in a process and it is sequential . However , I do not find any other possible reason . Finally thank you for your efforts
[image]

sergeykhbr, 2023-07-12:
I'm glad that you found solution but I do not see there any functional changes. Last commit removes double assignments into process variable path_id, if this helps with ISim let it be here.

mito200, 2023-07-13:
Yes in Xsim it was not working because of this bug

sergeykhbr, 2023-07-14:
LRU. See generic parameters of the Workgroup/RiverTop modules.