sergeykhbr/riscv_vhdl #48: Should SRLW be I Type instead of R Type?

dselke, 2024-10-23:
You have a great project here Sergey. I fixed a bug in my old version of River by switching SRLW from R Type ("ADD like") to I Type ("ADDI like"), and I just want to make sure you don't have the same issue.  I was getting a bogus shift amount in the second shifter argument in simulation which caused the River sim to disagree with the same .c file built for x64 laptop. Same instruction, shift right logical word by 31. Instead of this instruction turning "all FFs" into "1", the result in the sim was "00000000000003FF" (shift by 14 instead of 31 - the instruction in both .s files showed 31.  The problem was that src2 is supposed to be immediate  (https://riscv.org/wp-content/uploads/2019/06/riscv-spec.pdf, page 37).  On line 633 and 688 of [decoder.vhd](https://github.com/sergeykhbr/riscv_vhdl/blob/master/vhdl/rtl/riverlib/core/decoder.vhd), it looks like SRLW is still R Type.  I don't know if this is still a bug for you because my code diverged from your latest some years ago.

Live long and prosper! :)

dselke, 2024-10-23:
I have a custom assembler as well as a customized river cpu so this "fix" could be a case of two wrongs making a right!

sergeykhbr, 2024-10-24:
Hi,
VHDL code is a bit obsolete. Newer SystemVerilog/SystemC version is available and it was tested with third-party riscv-tests framework.
Could you share patch/diff file?

dselke, 2024-10-25:
I was confused reading the spec.  SRLW shift amount is supposed to be rs2[4:0], which I misinterpreted as the 5 bits labeled rs2 in the "diagram" of the instructions itself. Instead it is the 5 low bits of the register whose name is in that field of the instruction.  So it should be R-Type.  I am now sure that I fixed a software bug by adding a hardware bug. SRLIW does what I was trying to make SRLW do.  False alarm! There's no need for you to maintain obsolete stuff; I will own my copy of the vhdl until I can update to the latest river. Thanks for your quick response.

dselke, 2024-10-25:
Trying to close as resolved.

closed 2024-10-30 by sergeykhbr