sergeykhbr/riscv_vhdl #49: Help in simulating the file

jibinbabuamakkattu, 2025-02-14:
We have been trying to simulate the rivertop file. What we are doing is that, we are using force constant to assign values to each input variable.
We force constant all the required variables in the fetch stage to check if the output is available in the execute stage, but the o_valid is always 0 for which ever instruction we try.
Could you kindly help us understand what we are doing wrong?

sergeykhbr, 2025-02-15:
Sorry friend, there's not enough information. What target you are trying simulate? What is toolchain?

jibinbabuamakkattu, 2025-02-17:
Thanks for your reply. Here's some additional context:

Target & Toolchain: We're simulating the rivertop file using Xilinx Vivado 2023.2, with the intent to eventually implement it on an Artix-7 board.
Simulation Scope: We’re focusing solely on the rivertop file rather than simulating the complete repository.
Issue Encountered: In our simulation, we’re using force constant to assign values to all required input signals in the fetch stage. Despite this, the o_valid signal in the execute stage remains 0 for every instruction we test.
Could you help us understand what might be going wrong? Are there any dependencies or simulation setup issues (e.g., missing reset sequences, handshake signals, or other initialization conditions) that we might be overlooking by simulating just the top-level file with forced constants?

Any insights or suggestions would be greatly appreciated!